BCDA-APS/8id-bits #31: Scan mirror1.X while recording areadet flag1 hdf at each point

prjemian, 2024-06-10:


prjemian, 2024-06-10:
- refs #17

prjemian, 2024-06-10:
Expect this command should scan from 1 to 5 with 11 points, producing a single HDF5 file at each point:

```py
RE(bp.scan([flag1ad], mr1.x, 1, 5, 11))
```

But, might need some attention to how the area detector is configured before the scan.  Also, to get scalar values (total counts in ROI), need to identify and add that python object as additional detector.

prjemian, 2024-06-10:
```py
In [14]: mr1.x.position
Out[14]: 1540.3894

In [15]: mr1.x.limits
Out[15]: (1539.03575, 1549.03575)
```

Use different start and end positions, within the limits, to test the scan.  Such as 1542, 1545, 11

prjemian, 2024-06-10:
The names of the parts:

```py
In [16]: mr1.component_names
Out[16]: ('x', 'y', 'coarse_pitch', 'fine_pitch', 'flag')
```

qzhang234, 2024-11-18:
We have demonstrated successful 1D scan using `sample.x` and `eiger4M.stats.roi1_mean`. This issue can be marked as resolved.

closed 2024-11-18 by qzhang234