BCDA-APS/8id-bits #33: In each case, we will add a ROI and a Stats to plot the counter as well

prjemian, 2024-06-10:


prjemian, 2024-06-10:
? `flag1ad.stats1.total`

sureshnaps, 2024-06-11:
> ZjQcmQRYFpfptBannerStart
> This Message Is From an External Sender
> This message came from outside your organization.
>
> ZjQcmQRYFpfptBannerEnd

Pete,
I have been reviewing and the plans look very good.
When ready, we can run a scan.

One good question/comment you had was:
…Provided the areadet flag plugins are correctly set up.

This was in my mind as well. With such cameras, it might make sense to collect N (~10) images per data point in case one of them has any issues. We will have to software trigger the camera appropriately.

Suresh

prjemian, 2024-06-11:
Note: to record the `flag1ad.stats1.total` (or any other signal) as another detector, it needs to have its `.kind = "hinted"`, as in:

```py
flag1ad.stats1.total.kind = "hinted"
```

This is an internal (bluesky & ophyd) setting that is needed to report this signal when `flag1ad` is used as a detector..

prjemian, 2024-06-11:
Another signal you might be interested in: `flag1ad.stats1.max_value`

prjemian, 2024-06-11:
Now that the detector is connected:

```py
In [4]: flag1ad.stats1.total
Out[4]: EpicsSignalWithRBV(read_pv='8idaSoft:flag1:Stats1:Total_RBV', name='flag1ad_stats1_total', parent='flag1ad_stats1', timestamp=1718126235.5730493, auto_monitor=True, string=False, write_pv='8idaSoft:flag1:Stats1:Total', limits=False, put_complete=False)

In [5]: flag1ad.stats1.total.get()
Out[5]: 0.0

In [6]: flag1ad.stats1.total.kind
Out[6]: <Kind.omitted: 0>

In [7]: flag1ad.stats1.total.kind = "hinted"

In [8]: flag1ad.stats1.total.kind
Out[8]: <Kind.hinted: 5>

In [9]: flag1ad.read()
Out[9]: OrderedDict()

In [10]: flag1ad.stats1.read()
Out[10]: 
OrderedDict([('flag1ad_stats1_total',
              {'value': 0.0, 'timestamp': 1718126235.5749059})])

In [11]: flag1ad.stats1.kind
Out[11]: <Kind.omitted: 0>

In [12]: flag1ad.stats1.kind = "hinted"

In [13]: 

In [13]: flag1ad.read()
Out[13]: 
OrderedDict([('flag1ad_stats1_total',
              {'value': 0.0, 'timestamp': 1718126235.5749059})])
```

Summary:

```py
flag1ad.stats1.kind = "hinted"
flag1ad.stats1.total.kind = "hinted"
```

prjemian, 2024-06-11:
Here's an example notebook showing the stats info in a bluesky scan:
[roi1_detector.pdf](https://github.com/user-attachments/files/15792877/roi1_detector.pdf)

qzhang234, 2024-11-18:
We have demonstrated successful 1D scan using `sample.x` and `eiger4M.stats.roi1_mean`. This issue can be marked as resolved.

closed 2024-11-18 by qzhang234