BCDA-APS/8id-bits #60: create ophyd objects for certain beam line equipment

prjemian, 2024-08-27:
Create support for the items marked by red boxes:
[image]

- [ ] Transfocator in 8-ID-D & 8-ID-E
- [ ] Kirkpatrick-Baez Mirrors in 8-ID-E
- [ ] Huber diffractometer in 8-ID-E
- [ ] X-ray beam position monitor in 8-ID-E

prjemian, 2024-08-27:
The 8-ID-E and 8-ID-I stations are separate instruments.  Are we combining support for both in this one repository?

prjemian, 2024-08-27:
If they will be operated at the same time, we should not combine their controls here.

MDecarabas, 2024-08-27:
I think all of 8id can live in 1 package, and depending on which station it is being used on different devices can be imported

prjemian, 2024-08-27:
Asked the beam line team if the two instruments are operated at the same time.  That's the important point to make the decision.

sureshnaps, 2024-08-27:
> ZjQcmQRYFpfptBannerStart
> This Message Is From an External Sender
> This message came from outside your organization.
>
> ZjQcmQRYFpfptBannerEnd

the 2 instruments will not take beam at the same time but i can imagine operating the 2 instruments at the same time.

sureshnaps, 2024-08-27:
as long as I can run bluesky8IDI and bluesky8IDE concurrently and also 8IDI should not load all the 8IDE devices and vice versa unless needed by choice

prjemian, 2024-08-27:
Thanks.  That's a good requirement for us to meet.  @MDecarabas & I will work it out.

prjemian, 2024-08-28:
Issues are created for all objects.  Planning for concurrent operations of E & I stations will continue is separate issue.

closed 2024-08-28 by prjemian