BCDA-APS/8id-bits #65: run bluesky8IDI and bluesky8IDE concurrently

prjemian, 2024-08-27:
... as long as I can run bluesky8IDI and bluesky8IDE concurrently and also 8IDI should not load all the 8IDE devices and vice versa unless needed by choice

From: Suresh N. ***@***.***>
Date: Tuesday, August 27, 2024 at 2:24 PM
To: aps-8id-dys/bluesky ***@***.***>
Cc: Narayanan, Suresh ***@***.***>, Your activity ***@***.***>
Subject: Re: [aps-8id-dys/bluesky] create ophyd objects for certain beam line equipment (Issue #60)
the 2 instruments will not take beam at the same time but i can imagine operating the 2 instruments at the same time. ________________________________ From: Pete R Jemian ***@ ***. ***> Sent: Tuesday, August 27, 2024 1: 49 PM To: aps-8id-dys/bluesky
ZjQcmQRYFpfptBannerStart
This Message Is From an External Sender
This message came from outside your organization.

ZjQcmQRYFpfptBannerEnd
the 2 instruments will not take beam at the same time but i can imagine operating the 2 instruments at the same time.
________________________________
From: Pete R Jemian ***@***.***>
Sent: Tuesday, August 27, 2024 1:49 PM
To: aps-8id-dys/bluesky ***@***.***>
Cc: Subscribed ***@***.***>
Subject: Re: [aps-8id-dys/bluesky] create ophyd objects for certain beam line equipment (Issue #60)

Asked the beam line team if the two instruments are operated at the same time. That's the important point to make the decision. — Reply to this email directly, view it on GitHub, or unsubscribe. You are receiving this because you are subscribed
ZjQcmQRYFpfptBannerStart
This Message Is From an External Sender
This message came from outside your organization.

ZjQcmQRYFpfptBannerEnd

Asked the beam line team if the two instruments are operated at the same time. That's the important point to make the decision.

—
Reply to this email directly, view it on GitHub<https://urldefense.us/v3/__https://github.com/aps-8id-dys/bluesky/issues/60*issuecomment-2313277761__;Iw!!G_uCfscf7eWS!e9Ta48Ji1rMuSA4BDNWlNdOpdA5hyUfmLkyavP3JNTGmqulNhwLJuOdVP9WP9YC32kDZ8tzzwU9n7lKUNlk0bBZtmA$%3E, or unsubscribe<https://urldefense.us/v3/__https://github.com/notifications/unsubscribe-auth/AAYSAQ3AZ5HEYHV37VCNLO3ZTTC4NAVCNFSM6AAAAABNGWE466VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDGMJTGI3TONZWGE__;!!G_uCfscf7eWS!e9Ta48Ji1rMuSA4BDNWlNdOpdA5hyUfmLkyavP3JNTGmqulNhwLJuOdVP9WP9YC32kDZ8tzzwU9n7lKUNlnGy8HEXg$%3E.
You are receiving this because you are subscribed to this thread.Message ID: ***@***.***>

—
Reply to this email directly, view it on GitHub<https://urldefense.us/v3/__https:/github.com/aps-8id-dys/bluesky/issues/60*issuecomment-2313340601__;Iw!!G_uCfscf7eWS!YCM2srsGWhL4000oym7hw89dBFfQ1rbuLHaw65J-8tt8AfEp8Qmn5otl_j5Qc1awT0MarBW_qR14ANJyfAhHFnV2lQ$>, or unsubscribe<https://urldefense.us/v3/__https:/github.com/notifications/unsubscribe-auth/AAYSAQ7PUMB2S3B6O7X4R4LZTTG7RAVCNFSM6AAAAABNGWE466VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDGMJTGM2DANRQGE__;!!G_uCfscf7eWS!YCM2srsGWhL4000oym7hw89dBFfQ1rbuLHaw65J-8tt8AfEp8Qmn5otl_j5Qc1awT0MarBW_qR14ANJyfAi-9R5Uxg$>.
You are receiving this because you are subscribed to this thread.Message ID: ***@***.***>

_Originally posted by @sureshnaps in https://github.com/aps-8id-dys/bluesky/issues/60#issuecomment-2313386019_

prjemian, 2024-08-27:
Similar permissions were added to 2-BM for operations in either A or B station.  An EPICS mbbo record [defines](https://github.com/APS-2BM-MIC/ipython-user2bmb/issues/53#issuecomment-439650986) which station could control.

[image]

The [bluesky code](https://github.com/APS-2BM-MIC/ipython-user2bmb/issues/53#issuecomment-440079250) looked like:

```py
instrument_in_use = EpicsSignalRO("2bm:instrument_in_use", name="instrument_in_use", string=True)
MY_STATION = "2-BM-B

def using_2bmb(station):
    return instrument_in_use.get() == station

# ...

if using_2bmb(MY_STATION):
    # connect to real things
    # suspend when NOT using_2bmb
else:
    # connect to simulated things
    # suspend when using_2bmb
```

sureshnaps, 2024-08-27:
To be clear,
I can imagine doing x-ray expt in one station and a non x-ray in another and let us assume that both of these will run as if x-rays are there which means moving real motors, taking detector images, etc.

image.png (view on web)<https://urldefense.us/v3/__https:/github.com/user-attachments/assets/94d0cc22-df6f-4cdd-9315-55aef990bd14__;!!G_uCfscf7eWS!b1Rn8a6EQalD6Zq10p-GFRpOVLKpJP6wvK-zWhasNX-zyPpmmJ6B6yJ18yhqigpXDCntMtK3c2YZbkWkFtlVkSs7tQ$>