BCDA-APS/8id-bits #68: Check tetramm.summary() command

prjemian, 2024-08-30:
Root cause:

```py
In [50]: xbpm2.validate_asyn_ports()
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Cell In[50], line 1
----> 1 xbpm2.validate_asyn_ports()

File ~/.conda/envs/bluesky_2024_2/lib/python3.11/site-packages/ophyd/areadetector/base.py:465, in ADBase.validate_asyn_ports(self)
    463 if port_map and nx.number_connected_components(g) != 1:
    464     missing_plugins = self.missing_plugins()
--> 465     raise RuntimeError(
    466         "The asyn ports {!r} are used by plugins "
    467         "that ophyd is aware of but the source plugin "
    468         "is not.  Please reconfigure your device to "
    469         "include the source plugin or reconfigure "
    470         "to not use these ports."
    471         "".format(missing_plugins)
    472     )

RuntimeError: The asyn ports ['T4U_BPM'] are used by plugins that ophyd is aware of but the source plugin is not.  Please reconfigure your device to include the source plugin or reconfigure to not use these ports.

```

Check the TetrAMM for similar.

_Originally posted by @prjemian in https://github.com/aps-8id-dys/bluesky/issues/64#issuecomment-2322189594_

qzhang234, 2024-10-07:
This bug has been fixed

closed 2024-10-07 by qzhang234

prjemian, 2024-10-07:
How?

qzhang234, 2024-10-07:
> ZjQcmQRYFpfptBannerStart
> This Message Is From an External Sender
> This message came from outside your organization.
>
> ZjQcmQRYFpfptBannerEnd

I noticed that tetramm1.summary() return the right list of attributes when I was trying to scan with it last Sunday (10/06). The scan didn't work, but I was able to read the current and all other attributes.

QZ