BCDA-APS/8id-bits #77: bp.rel_scan does not trigger eiger4M

qzhang234, 2024-09-26:
**Describe the bug**
bp.rel_scan does not trigger eiger4M

**To Reproduce**
I changed `eiger4M.stats1.mean_value` to `hinted` and did:
```
RE(bp.rel_scan([eiger4M.stats1.mean_value],sample.x, -2, 2, 20))
```

**Expected behavior**
I expect the scan to trigger the detector at every step, and I'm able to view the spec file with pyMca.

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
 - OS: Linux RHEL
 - Environment: bluesky_2024_3, Code: `/home/beams/8IDIUSER/bluesky/scripts`

**Additional context**
`bp.count` works and can trigger the detector
[image]

prjemian, 2024-09-27:
Can you repeat the same scan with this one (seemingly) minor change?
`RE(bp.rel_scan([eiger4M],sample.x, -2, 2, 20))`

### Explanation

Your previous command, `RE(bp.rel_scan([eiger4M.stats1.mean_value],sample.x, -2, 2, 20))` seemed like the right idea.  You asked for the signal you want to see.  But this signal does not know how to trigger its parent device.  So instead, you *configure* (as you have done) the signal to be reported.  Then you supply the entire detector to the `count` plan.  The detector knows how to trigger.

prjemian, 2024-09-27:
That did it!

closed 2024-09-27 by prjemian