BCDA-APS/8id-bits #83: Can't scan tetramm in Bluesky

qzhang234, 2024-10-07:
I'm trying to scan tetramm3 with sl4. The command I entered is:
```
RE(bp.rel_scan([tetramm3],sl4.h.center, -3, 3, 60))
```
I got an error message that says: 
```
RuntimeError: The asyn ports ['QUAD3'] are used by plugins that ophyd is aware of but the source plugin is not.  Please reconfigure your device to include the source plugin or reconfigure to not use these ports.
```
[image]

prjemian, 2024-10-07:
An error report like that coming from a TetrAMM (or other ophyd device based on ophyd.area_detector) usually means that the device is not created properly or the EPICS IOC area detector plugin ports are different than expected.

Ophyd has some diagnostic code to use.  This reports the area detector ports in use and what ophyd class is used for each plugin.

```py
In [3]: tetramm3.get_asyn_port_dictionary()
Out[3]: 
{'{port_name}': QuadEMPort(prefix='', name='tetramm3_conf', parent='tetramm3', read_attrs=[], configuration_attrs=[]),
 'QUAD3_Image1': ImagePlugin_V34(prefix='8idTetra:QUAD3:image1:', name='tetramm3_image', parent='tetramm3', read_attrs=[], configuration_attrs=[]),
 'QUAD3_STATS1': StatsPlugin_V34(prefix='8idTetra:QUAD3:Current1:', name='tetramm3_current1', parent='tetramm3', read_attrs=['mean_value'], configuration_attrs=[]),
 'QUAD3_STATS2': StatsPlugin_V34(prefix='8idTetra:QUAD3:Current2:', name='tetramm3_current2', parent='tetramm3', read_attrs=['mean_value'], configuration_attrs=[]),
 'QUAD3_STATS3': StatsPlugin_V34(prefix='8idTetra:QUAD3:Current3:', name='tetramm3_current3', parent='tetramm3', read_attrs=['mean_value'], configuration_attrs=[]),
 'QUAD3_STATS4': StatsPlugin_V34(prefix='8idTetra:QUAD3:Current4:', name='tetramm3_current4', parent='tetramm3', read_attrs=['mean_value'], configuration_attrs=[]),
 'QUAD3_STATS7': StatsPlugin_V34(prefix='8idTetra:QUAD3:SumAll:', name='tetramm3_sum_all', parent='tetramm3', read_attrs=['mean_value'], configuration_attrs=[])}

```

prjemian, 2024-10-07:
This looks odd: `{'{port_name}': ...`  I'm expecting to see `{'{QUAD3}': ...`  

I'm looking for a typo in the source code.  This looks like a Python _f_-string is intended but the `f` is missing.

prjemian, 2024-10-07:
See the problem here? https://github.com/aps-8id-dys/bluesky/blob/a8b8fda4ad62d20f26b8a6a4de6b5a3be06a97cc/src/instrument/devices/tetramm_picoammeter.py#L25

prjemian, 2024-10-07:
Should be

```py
 conf = FormattedComponent(QuadEMPort, port_name="f{port_name}") 
```

prjemian, 2024-10-07:
Not quite the solution.  It's using an ophyd `FormattedComponent` so the correction needs a bit of research to get it right.

prjemian, 2024-10-07:
Looks like this now:

```py
In [16]: tetramm3.get_asyn_port_dictionary()
Out[16]: 
{'QUAD3': QuadEMPort(prefix='', name='tetramm3_conf', parent='tetramm3', read_attrs=[], configuration_attrs=[]),
 'QUAD3_Image1': ImagePlugin_V34(prefix='8idTetra:QUAD3:image1:', name='tetramm3_image', parent='tetramm3', read_attrs=[], configuration_attrs=[]),
 'QUAD3_STATS1': StatsPlugin_V34(prefix='8idTetra:QUAD3:Current1:', name='tetramm3_current1', parent='tetramm3', read_attrs=['mean_value'], configuration_attrs=[]),
 'QUAD3_STATS2': StatsPlugin_V34(prefix='8idTetra:QUAD3:Current2:', name='tetramm3_current2', parent='tetramm3', read_attrs=['mean_value'], configuration_attrs=[]),
 'QUAD3_STATS3': StatsPlugin_V34(prefix='8idTetra:QUAD3:Current3:', name='tetramm3_current3', parent='tetramm3', read_attrs=['mean_value'], configuration_attrs=[]),
 'QUAD3_STATS4': StatsPlugin_V34(prefix='8idTetra:QUAD3:Current4:', name='tetramm3_current4', parent='tetramm3', read_attrs=['mean_value'], configuration_attrs=[]),
 'QUAD3_STATS7': StatsPlugin_V34(prefix='8idTetra:QUAD3:SumAll:', name='tetramm3_sum_all', parent='tetramm3', read_attrs=['mean_value'], configuration_attrs=[])}

```

prjemian, 2024-10-07:
Apparently, using a templated string with `FormattedComponent` only works for the PV name part, not for a keyword attribute such as `port_name`.

commit 21034136 (2024-10-07): FIX #83 put port name to component
closed 2024-10-07 by prjemian