BCDA-APS/8id-bits #89: main branch was not reset properly

prjemian, 2024-10-18:
On 2024-10-17, a PR was merged (66c5a7a) before its changes were tested.  That merge was reverted (812eb1e) and then another correction applied (07b7f19).  

[image]

These changes caused a merge conflict with existing branch [`81-xpcs_mesh+DM`](https://github.com/aps-8id-dys/bluesky/tree/81-xpcs_mesh%2BDM) such that changes in `main` would remove substantial new changes in the new branch.

The `81-xpcs_mesh+DM` branch was last synchronized with `main` branch commit 0ae10c6.

[image]

prjemian, 2024-10-18:
One suggestion was to drop the existing main branch and switch the new one to be main.  That idea has problems, too.  The beamline needs to continue its work.  I believe now the `main` branch should be repaired.

- No new commits should be made to the main branch until the problem is resolved.
- Recent commits to `main` after 0ae10c6 would need to be recommitted.
  - A new branch for other work (https://github.com/aps-8id-dys/bluesky/tree/individual_slits) has been created.
  - 7a3d38b (increase default slit size tolerance)
- The beamline needs to continue its work with the main branch.

prjemian, 2024-10-18:
### references

- https://stackoverflow.com/a/44975300
  > Another alternative which worked for me was to reset the Git head and index to its previous state using: `git reset --keep`
- https://git-scm.com/book/en/v2/Git-Basics-Undoing-Things
- https://stackoverflow.com/questions/4830056/how-to-discard-all-changes-made-to-a-branch

prjemian, 2024-10-18:
I'll explore the `git reset --keep` in an offline directory and report results here.

prjemian, 2024-10-18:
Comparing the previous _good_ commit of main (0ae10c6) with the current one (7a3d38b96),  the slit size tolerance is changed:

```bash
git diff 0ae10c6 7a3d38b96
diff --git a/src/instrument/devices/slit.py b/src/instrument/devices/slit.py
index 0ec6a02..a97efd0 100644
--- a/src/instrument/devices/slit.py
+++ b/src/instrument/devices/slit.py
@@ -42,7 +42,7 @@ class ID8Optics2Slit1D(Optics2Slit1D):
         "size",
         setpoint_pv=".VAL",
         readback_pv=".RBV",
-        tolerance=8e-4,
+        tolerance=2e-3,
     )
     center = cpt(
         PVPositionerSoftDone,
```

prjemian, 2024-10-18:
Concluding that the problems merging main into https://github.com/aps-8id-dys/bluesky/tree/81-xpcs_mesh%2BDM is because some main branch file timestamps are newer.

#### Summary

Problem is in main branch.  Since it is obviously not ready now and basing anything else on it could pose more problems, I'll save the changes in my branch and start a new branch.  That would be less burden on us all.  @qzhang234  wants a new feature anyway.

MDecarabas, 2024-10-18:
Sounds good

> Concluding that the problems merging main into https://github.com/aps-8id-dys/bluesky/tree/81-xpcs_mesh%2BDM is because some main branch file timestamps are newer.
>
> #### Summary
> Problem is in main [...]

Sounds good!

closed 2024-10-18 by MDecarabas

prjemian, 2024-10-18:
> Problem is in main my branch.

Should be: 
Problem is in `81-xpcs_mesh+DM` branch.