BCDA-APS/8id-bits #104: Independent function for starting the DM workflow after acquisition and nexus file writing is complete

qzhang234, 2024-11-04:
We would like to have an independent function for starting the DM workflow after the acquisition is done and the nexus hdf metadata file is written. 

We would prefer to have a plain python function independent from Bluesky RE (and potentially apstools, if possible) so that it is easier for beamline folks to develop, maintain and debug.

qzhang234, 2024-11-05:
@banubot provided a template for starting DM from the DM's python environment. Looks like the only input variables needed are in `argsDict` and there are only 3 of them. Would Bluesky be able to communicate these variables with Hannah's DM environment, either via text files or String Registers?

```
# This script is an example of how to start 
# the XPCS boost corr workflow using the DM python API
# Usage: source /home/dm/etc/dm.setup.sh; python dm_api_example.py

from dm.proc_web_service.api.procApiFactory import ProcApiFactory

# Get the DM workflow API object
dmWorkflowApi = ProcApiFactory.getWorkflowProcApi()

# Define the workflow arguments in a dict
argsDict = {
    "filePath": "H001_005_test_Feb_7-01000.h5",
    "qmap": "eiger4M_qmap_d36_s360.h5",
    "experimentName": "zhang202402",
}

workflowOwner = "8idiuser"
workflowName = "xpcs8-02-gladier-boost"
# Start the processing job
response = dmWorkflowApi.startProcessingJob(workflowOwner, workflowName, argsDict)
# Print the job ID
print(response.get('id'))

# To do the same thing using CLI instead of python, run:
# dm-start-processing-job --workflow-name xpcs8-02-gladier-boost filePath:H001_005_test_Feb_7-01000.h5 qmap:eiger4M_qmap_d36_s360.h5 experimentName:zhang202402
```

prjemian, 2024-11-06:
Bluesky has a different set of Python package requirements than APS Data Management.  The bluesky instrument package has provided an adaptation that provides access to the full set of Python software provided by APS DM.  The adaptation is enabled by defining all the environment variables required by DM:
 https://github.com/aps-8id-dys/bluesky/blob/6bbcfeceab7a6695d3be81ffd56954d362bf25ea/src/instrument/utils/aps_data_management.py#L272

Various DM APIs are already provided: https://github.com/aps-8id-dys/bluesky/blob/6bbcfeceab7a6695d3be81ffd56954d362bf25ea/src/instrument/utils/aps_data_management.py#L8-L14

For example, `dm_api_cat`: https://github.com/aps-8id-dys/bluesky/blob/6bbcfeceab7a6695d3be81ffd56954d362bf25ea/src/instrument/utils/aps_data_management.py#L148-L153

prjemian, 2024-11-06:
Since `dm_source_environ()` is run when starting a bluesky session, @banubot's example above can be run from a bluesky session.

prjemian, 2024-11-06:
@qzhang234 Is there any additional work to be done to resolve this issue?

qzhang234, 2024-11-06:
> ZjQcmQRYFpfptBannerStart
> This Message Is From an External Sender
> This message came from outside your organization.
>
> ZjQcmQRYFpfptBannerEnd

Hi Pete:

Allow me to do some more research. In the meantime please keep the issue open.

Thank you,

QZ

prjemian, 2024-11-20:
A low-level plan (bluesky plan stub that wraps DMs `startProcessingJob()` function) is described in the [`bs_model_instrument`](https://github.com/BCDA-APS/bs_model_instrument/blob/319c4947dd917f7a85772e758f337366a358ce5b/docs/source/guides/dm.md?plain=1#L65).  This is part of PR https://github.com/BCDA-APS/bs_model_instrument/pull/20.