BCDA-APS/8id-bits #106: Name of metadata file stored as array of int

prjemian, 2024-11-05:
The name of the NeXus HDF5 metadata file is stored in the run's metadata.  Turns out it is stored as a list of `int` rather than a text string.  

**To Reproduce**
Here's a view (from GemViz)  of a recent XPCS run:
[image]

**Expected behavior**
This example should be: `'/gdata/dm/8IDI/2024-3/foster202410/data/E764_E-BL-3_a0010_t1299_f000480_r00001/E764_E-BL-3_a0010_t1299_f000480_r00001.hdf'`

prjemian, 2024-11-05:
Convert this key to `str`:

`filename = "".join([chr(int(c)) for c in metadata if int(c) > 0])`

prjemian, 2024-11-05:
Was not a problem in run `a1285dc5-4be6-4ded-93b7-b9c40602fe90` from 2024-10-28, file `/gdata/dm/8IDI/2024-3/comm202410/data/A001_001/A001_001.hdf`.

[image]

prjemian, 2024-11-05:
`type(metadata)` | date | uid | filename
---- | ---- | ---- | ----
`str` | 2024-10-28 17:20 | `aa6b4c99-4551-4010-8fd9-bda95b57c81a` | `/gdata/dm/8IDI/2024-3/comm202410/data/B003_004/B003_004.hdf`
`list[int]` | 2024-10-29 17:06 | `776c26a5-dd18-4320-8215-32a4ce49d22e` | `xpcs8-02-gladier-boost`

These are consecutive runs.

prjemian, 2024-11-05:
From the `git log` history, check these commits:

- 18be42405466a2d55f4314fee80be58217c85b25 (oregistry)
- d92aebffa36ce17847d903d77a1ac3201daac5a0 (softglue)
- a7cf00480c8016b575d23547d94f1dcd903dae3c

In commit a7cf00480c8, here's where the bug was introduced:

[image]

and where the previous code was removed:

[image]

prjemian, 2024-11-05:
With the work of #103, will this `metadata` key be needed?

prjemian, 2024-11-05:
"Name of metadata file", such as: `/gdata/dm/8IDI/2024-3/comm202410/data/B003_004/B003_004.hdf`

qzhang234, 2024-11-05:
Good point. I believe this does not hurt to have, but if it's too much trouble then we'll skip it

prjemian, 2024-11-05:
That's not really the point.

When creation of a metadata file and start of a DM workflow are part of the bluesky plan, the *name* of the metadata file is added to the bluesky run's metadata.  In that case, the complete dictionary of the run's metadata (including the metadata file name) is copied to the DM experiment for cross-reference.

Given issue #104, not only is it not important to write the metadata file name, it's not possible since the name is not yet known.

prjemian, 2024-11-05:
If the current DM workflows are executing with complete success, we should delete this key.  Such success is obvious indication it is not used.

qzhang234, 2024-11-05:
Yes, I agree. Please go ahead with the deletion. Thanks!

prjemian, 2024-11-06:
Also note, when using PyEpics with the string register PVs, add the `as_string=True` kwarg, such as this example:

```py
In [1]: import epics as pe

In [2]: pe.caget("8idi:StrReg27")
Out[2]: 
array([120, 112,  99, 115,  56,  45,  48,  50,  45, 103, 108,  97, 100,
       105, 101, 114,  45,  98, 111, 111, 115, 116,   0], dtype=uint8)

In [3]: pe.caget("8idi:StrReg27", as_string=True)
Out[3]: 'xpcs8-02-gladier-boost'

```

prjemian, 2024-11-06:
Closing since this code will be superseded by #107.

closed 2024-11-06 by prjemian