BCDA-APS/8id-bits #109: Add Ophyd device for attenuations in 8-ID

qzhang234, 2024-11-18:
There are two of them. 

The one at 8-ID-E is: `8idPyFilter:FL2:sortedIndex`
The one at 8-ID-I is: `8idPyFilter:FL3:sortedIndex`

Max Wyman set up the softIOCs for these two attenuators.

qzhang234, 2024-11-18:
This is the same AVS-US filter from 12-ID, and quite a few other beamlines have the exact same unit. @prjemian I would like to see if there has already been an established template for it. If not, I would like to take the first shot in creating this

prjemian, 2024-11-20:
There is code to address this in PR #101.

prjemian, 2024-11-21:
@QZ wrote in MS Teams:

> For the filter, I was wondering if you could try a simpler definition. We have two identical attenuator in 8-ID-E and 8-ID-I. Each of them has three boxes we can control: the attenuation, the att index, and the transmission. If we specify the desired attenuating factor or the transmission ratio (e.g., 5 times attenuation or 0.2 transmission), Max's wonderful EPICS software will put the attenuation level to the closest number calculated from the software. This also means that the readback value will not be exactly the same as the set value. However, if we change the attenuation index (which is always an integer), the readback value will match the set value, but we will not know how much attenuation it corresponds to.
> 
> Given the unique design of the attenuator box, I think we could just control the att index from Bluesky, and put the attenuation/transmission readback into metadata. Can we remove `filter_enable_signal` and `filter_op.py` and make a simple device that has one EpicsSignal (atte index) and two EpicsSignalRO (attenuation and transmission)?

qzhang234, 2024-11-21:
A minimalist implementation was implemented and tested on 11/21. The issue can be marked as solved.

[image]

closed 2024-11-21 by qzhang234