BCDA-APS/8id-bits #112: Simple python code that converts Data Broker Dump to text files with Spec format

qzhang234, 2025-03-14:
I would like to have a code that converts the contents dumped from the Data Broker to something with a Spec structure. Since both the DB dump and the Spec file are structured text files, I imagine this can be achieved with minimal uses of libraries. 

A source code exists that writes the spec file line-by-line when new events occur. Would be nice to have something similar to this code but with much simpler structure since the writing needs to happen only once after the scan finishes.

@prjemian Please advise on how I should proceed.

qzhang234, 2025-03-14:
Comments from Pete: Pre-requisite for this request:

1. structure of the data available from db[-1] and how to re-arrange it for SPEC
2. structure of the descriptors in db[-1]
3. structure of the data in SPEC's format
4. how to add new scans to existing SPEC files
5. how and when to add a header section in a SPEC file

prjemian, 2025-10-31:
That's the right documentation.  Click the "[source]" link to see the Python source code.